Aceinna/python-openimu #1: Some browser can't access ws protocol under https

m5horton, 2018-06-25:
Example is firefox

m5horton, 2018-06-25:
This fixes firefox for this issue:
https://stackoverflow.com/questions/11768221/firefox-websocket-security-issue

fengliu00, 2018-07-30:
The solution for firefox: about:config->click conntinue->network.websocket.allowInsecureFromHTTPS->double click

fengliu00, 2018-07-30:
Developers.aceinna.com doesn't support Safari.

[image]

yiweisong, 2019-09-26:
I added a solution for Firefox and Safari.
The site will use wss protocol to listen message from python-openimu. And user should implement a websocket server that supports wss.

There is a tool about locally-trusted certificates, it helps to setup local wss server.
https://github.com/FiloSottile/mkcert

m5horton, 2019-12-17:
Resolved for all except Safari.

closed 2019-12-17 by m5horton