Aceinna/python-openimu #23: Python Ubuntu driver

FarhadMir, 2019-07-24:
Neither cloning the python driver from github nor installing through pip works for Ubuntu 16.04; it can search the ports but not connect to any.
[image]

yifanff, 2019-07-25:
Hi FarhadMir，what is the python version and pip version you are using?  Is the USB interface from IMU mount on the ubuntu system successfully?  what is the ‘lsusb’ command feedback？

FarhadMir, 2019-07-25:
Hey, Yifan; python 2.7, pip 19.2.1 (but I tried previous versions too); and yes it is succefully mounted look at the screen below:
[image]

yifanff, 2019-07-28:
Hi FarhadMir, thanks for your feedback! please git pull the code and make sure the IMU serial port have full access right, in my side: sudo chmod -R 777 /dev/ttyUSB0 , then it work normally.

FarhadMir, 2019-07-30:
Thanks, solved.

closed 2019-07-30 by FarhadMir

erkuo521, 2019-08-01:
at last what is the root cause in your side? based on your situation

erkuo521, 2019-08-01:
@FarhadMir at last what is the root cause in your side? based on your situation, cable, system not found divice firstly, account access right.......

FarhadMir, 2019-08-06:
account access right

erkuo521, 2019-08-07:
so, use command: "sudo usermod -a -G yourusergroup youraccountname" improve the account right will figure out the serial port cannot open problem