Aceinna/python-openimu #26: Issue with RPi4, /dev/ttyAMA0 vs /dev/serial0

SinginAlong, 2019-10-21:
Had trouble connecting to a OpenIMU300ZA with a Raspberry Pi 4 over serial.  The program would hang when trying to send data over the /dev/ttyAMA0 port.  I resolved the issue by hard-coding it to only try "/dev/serial0".

Setup info:  Using raspberry pi 4 with serial enabled and bluetooth turned off.  Connected to IMU UART0 lines from RPi GPIO 14 & 15.

sulkytejas, 2019-10-22:
@SinginAlong I think issue here might be the way you have enabled the port. For Pi 3 or later there is one additional step to enable serial port. 
1. Open config.txt
2. add enable_uart=1
3. sudo reboot
4. launch raspi-config disable console and keep serial enabled 

Just make sure your USB is on ttyS* instead of ttyMA. Please have a look at this document https://www.raspberrypi.org/documentation/configuration/uart.md  for more info on UART on Pi. 
Please let me know if you have more questions.

SinginAlong, 2019-10-22:
Okay.  I had 1-3.  The console was not disabled.  I will try that later this week.  Thanks!

closed 2019-10-22 by SinginAlong