Aceinna/python-openimu #37: Bootloader code may sent the CRC in a wrong way (?)

carmelodfr, 2020-03-02:
Hi, I have an openimu300RI P/N 8350-3309-01 that has an inconsistent firmware. I am trying to flash the firmware by using the bootloader.py. However, I see the following problems:

a) The code does not send 'JI' packets continuosly. I modified the code to send every 10ms in order to enter the bootloader within the first 100ms. Also, I have to force ttyUSB0 because the device is inconsistent and does not communicate.

b) To be able to get into the bootloader I had to modify the script and send the CRC in the reversed order (first lsb ,then msb. Not the opposite like in the bootloader.py). Is the CRC sent in the correct order? Also the JA packet requires a CRC in the opposite order. This is problematic since when I send WA packets i do not receive an answer. I tried to reverse the CRC also in the WA, but i get no answer. I noticed that the CRC is computed on the buffer C[2: C[4]+5]. Shouldn't be C[4]+4 ?
Do you have any clue why this openimu requires a different packet format?

thanks in advance,

Carmelo

yiweisong, 2020-03-04:
I don't know about the details of packet format, but I can supply some ideas when you use bootloader.py.
About the usage of bootloader.py. You could run `python bootloader.py <path/to/your_firmware.bin>`.
If your imu device is under bootloader mode now, you could comment line 1295 `unit.start_bootloader()` to skip jump to bootloader, and run bootloader.py again. Then wait for the command finish.

carmelodfr, 2020-03-04:
Thank you for your reply.

My problem is that write_block() does not receive any packet (line 567). I suspect this is due to a wrong CRC computation  and the packet is discarded by the imu.  Note that, I had to modify the CRC order (msb/lsb) for the packet used to enter the bootloader