Aceinna/python-openimu #38: Find Port but Cannot Connect (for device openimu300ri)

kevindean, 2020-04-25:
Hello, I have been trying to get this working for a little while (as I'm new to this, it just may be something that I am doing incorrectly); however when following the README according to this github with ubuntu 18.04 (or 16.04, I get the same result so far), the serial port never seems to become connected. The device is visible, printing to the screen the /dev/ name ttyUSB0, but keeps trying to find it. I have changed the permissions to the port as well as recloning/pulling etc.  this is my printout from webserver:py:

Find device 0 times
system ports detected ['/dev/ttyUSB0']
Find device 1 times
system ports detected ['/dev/ttyUSB0']
Find device 2 times
system ports detected ['/dev/ttyUSB0']
Find device 3 times
system ports detected ['/dev/ttyUSB0']
Find device 4 times
system ports detected ['/dev/ttyUSB0']
Find device 5 times
system ports detected ['/dev/ttyUSB0']
Find device 6 times
system ports detected ['/dev/ttyUSB0']
Find device 7 times
system ports detected ['/dev/ttyUSB0']
Find device 8 times
system ports detected ['/dev/ttyUSB0']
Find device 9 times
system ports detected ['/dev/ttyUSB0']
Find device 10 times
system ports detected ['/dev/ttyUSB0']
Find device 11 times
system ports detected ['/dev/ttyUSB0']
Find device 12 times
system ports detected ['/dev/ttyUSB0']
Find device 13 times
system ports detected ['/dev/ttyUSB0']
Find device 14 times
system ports detected ['/dev/ttyUSB0']
Find device 15 times
system ports detected ['/dev/ttyUSB0']

I also tried loading this as a module in IPython to see what was happening, but it still wasn't connecting: same results;
In [52]: imu.find_ports()

system ports detected ['/dev/ttyUSB0']
Out[52]: []

In [53]: imu.connect()
Find device 0 times

system ports detected ['/dev/ttyUSB0']
Find device 1 times

system ports detected ['/dev/ttyUSB0']
Find device 2 times

system ports detected ['/dev/ttyUSB0']
Find device 3 times

system ports detected ['/dev/ttyUSB0']
Find device 4 times

system ports detected ['/dev/ttyUSB0']
Find device 5 times

system ports detected ['/dev/ttyUSB0']
Find device 6 times

system ports detected ['/dev/ttyUSB0']
Find device 7 times

system ports detected ['/dev/ttyUSB0']
Find device 8 times

system ports detected ['/dev/ttyUSB0']
Find device 9 times

system ports detected ['/dev/ttyUSB0']
Find device 10 times

system ports detected ['/dev/ttyUSB0']
Find device 11 times

system ports detected ['/dev/ttyUSB0']
Find device 12 times

system ports detected ['/dev/ttyUSB0']
Find device 13 times

system ports detected ['/dev/ttyUSB0']

do you have any advice on what to do next? thank you for your time.

yiweisong, 2020-04-26:
Please check if there is log information in path /{username}/openimu/webserver.log. The driver will log some run-time information. Please run the driver again, then find the log and take a look.

kevindean, 2020-04-27:
Here is the webserver log:

[webserver.log](https://github.com/Aceinna/python-openimu/files/4540235/webserver.log)

yiweisong, 2020-04-28:
I have read the log. The log is a little simple. It seems failed while getting device information. Could you start the driver with command `python3 webserver.py -l=20`, it could help to log more detailed information.
By the way, do you know the firmware version of your device?

closed 2020-05-13 by yiweisong