Aceinna/python-openimu #39: openrtk330无法在OpenRTK Monitor 上显示数据

SwingW, 2020-05-02:
[image]
Errors in running ans-de'vices
[image]

yiweisong, 2020-05-06:
It looks like the device is sending an empty packet.
1. Which version of driver are you using? Develop branch or an executor ?
2. Does it still happen after restart the driver?

SwingW, 2020-05-06:
[image]
Yes, restarting the program still does not work

yiweisong, 2020-05-06:
It seems your device is not the latest version. Could you contact the person who sell the device to you?

SwingW, 2020-05-06:
I don't quite understand what it means that the device is not the latest version? Do I need to update the firmware

SwingW, 2020-05-06:
I don't quite understand what it means that the device is not the latest version? How should I see the version of openrtk330

SwingW, 2020-05-06:
I don't quite understand what it means that the device is not the latest version? How should I see the version of openrtk330Li and how can i solve this problem

dw-gnss-aceinna, 2020-05-06:
@SwingW we had a firmware upgrade to a new architecture, please share us with ur email address  if it's ok, we can share you steps to upgrade your device first.

SwingW, 2020-05-06:
Ok, my mailbox is: 2460613245@qq.com,thank you

dw-gnss-aceinna, 2020-05-07:
email sent, any more questions let us know,

SwingW, 2020-05-07:
Hello, I followed the tutorial to update the firmware, and it also shows that the update was successful, why is my firmware version still 1.0.7
[image]
[image]

dw-gnss-aceinna, 2020-05-07:
firmware version is 1.0.7, correct, and the RTK_INS App is v1.2.2.

SwingW, 2020-05-07:
Yes, but no data is generated in the openrtk or openimu Monitor window, do you know what caused this situation?
[image]

dw-gnss-aceinna, 2020-05-07:
pls refer to online doc https://openrtk.readthedocs.io/en/latest/useOpenRTK.html for how-to-use

SwingW, 2020-05-10:
> 请参考在线文档https://openrtk.readthedocs.io/en/latest/useOpenRTK.html，了解如何使用

Hello, I am operating according to the online documentation tutorial, but I still have not received the gnss signal and imu data, and the open rtk330EVB does not blink when the power is connected to the yellow LED and the green LED. I want to ask you know what is the reason for this Is it

dw-gnss-aceinna, 2020-05-11:
Did u upgraded the firmware I sent u in the email? Or could you share us your WeChat, our engineer will help with it.

Best,

Da

SwingW, 2020-05-11:
My WeChat account is：18859275931，thank you for your help

closed 2020-11-26 by yiweisong