Aceinna/python-openimu #40: OpenIMU Connected, but Website says discconnected

kevindean, 2020-05-11:
Hello again,

I have to IMU connected to the port, however, the website says that the imu is not connnected. Advice? Thank you.

[image]

yifanff, 2020-05-12:
Hi, switch to chrome browser and refresh the develper page,please have a try.

kevindean, 2020-05-12:
Sounds good, I will give that a go and let you know the results. Thank you.

-Kevin

baweiji, 2020-05-12:
The web has a retry policy, if the web cannot detect the device in 50 times try, it will refuse to detect. You can try one of below actions,
1. Please refresh the page, and try it again. 
2. Click the chain icon on the right top of the page, then click the refresh button to detect device again.

closed 2020-05-27 by yiweisong