Aceinna/python-openimu #45: Openrtk300l

saraswathi-sirikonda, 2020-11-06:
Hello sir, I am very new to the research on  GNSS/INS integration.
Is openrtk300l provides Raw data of GNSS like pseudo ranges, range rates, position and velocity etc., for loosely coupled and tightly coupled integration.
Is it provides IMU raw data for integration.

dw-gnss-aceinna, 2020-11-09:
@saraswathi-sirikonda yes, it supports GNSS raw data in RTCM format and raw IMU data in binary format (which can then be decoded by this python driver to csv/text file); it also supports loosely integration as OpenRTK330LI outputs GNSS solution as well, please refer to _**openrtk.readthedocs.io**_ for more information

saraswathi-sirikonda, 2020-11-10:
Thank you for response,  I am interested to research on NavIC(IRNSS) also,  hence is provides raw data for NavIC signals.

saraswathi-sirikonda, 2020-11-10:
Thank you for your response, I am interested to research on NavIC(IRNSS) also, hence is OpenRTK330l provides raw data for NavIC signals?

dw-gnss-aceinna, 2020-11-10:
unfortunately it does not support IRNSS, but supports dual-frequency plan (L1/L5 or L1/L2) from the other four constellations (GPS, GLONASS, GALILEO, Beidou).

closed 2020-11-27 by dw-gnss-aceinna