Aceinna/python-openimu #71: How to change nmea data log frequency in web or pythondriver-win?

fei4xu, 2021-08-18:
Hi,

I'm using the downloaded pythondriver-win.zip from https://github.com/Aceinna/python-openimu/releases/tag/v2.5.2ict

It's working fine and all the yellow/green/red leds are flashing during the data logging.
I also used the tool with `parse` subcommand to parse the saved .bin into nmea log.

but I'd like to check how to change the nmea data log frequency? I have PC/web based configuration like this:
[image]
which parameter should I change?

I also have the Android tool, there's a clear setting page to change that value, can I use android tool for nmea freqency and then use PC pythondriver-win to log data?

yiweisong, 2021-08-19:
Firstly, please don't use the release with version v2.5.2ict, it is for specified user. It is better to use this one https://github.com/Aceinna/python-openimu/releases/tag/v2.5.0

yiweisong, 2021-08-19:
We don't supply nmea log frequency configuration now.

fei4xu, 2021-08-24:
[image]
This is the screenshot I took from the Android BLE App, on which `UserPacketRate` is configurable and the value is 100.
Can I use the Android App to configure it?

yiweisong, 2021-08-24:
I think so, you can try. The packet rate options you could refer to the RAWIMU in the web page.

closed 2021-08-25 by yiweisong