Aceinna/python-openimu #85: Change orientation of the OpenIMU axis

Ahmad-Drak, 2022-09-29:
using: OpenIMU300ZI, VG_AHRS app, 200Hz, a2 packet type, python driver (and ROS).

Looking at the OpenIMU driver settings for the VG_AHRS app, it strictly uses the +X+Y+Z axis orientation as indicated here:
https://github.com/Aceinna/python-openimu/blob/9aa2aab967e7328320053ab335d285415fe49e31/src/aceinna/setting/OpenIMU300ZI/VG_AHRS/openimu.json#L13

However for my application I need to rotate the IMU 90 degrees around the +Y axis. Therefore the desired orientation im testing with is -Z+Y+X. I updated the firmware to the -Z+Y+X orientation.

Since the orientation changes, wouldnt the order of the reading of the packets change as well? Where can I edit that?
(This can be potentiolly related to this: https://github.com/ROS-Aceinna/ros_openimu/blob/f0db402401c28102bb7ffda23b6f8a577d44471f/scripts/openimu_driver.py#L128 )

yiweisong, 2022-11-09:
No need change.

ysong-aceinna, 2022-11-09:
Hi @Ahmad-Drak ,
As you have updated desired orientation [-Z+Y+X], the order of the reading of the packets change as well. All accel and rate reading are in [-Z+Y+X]. So no need to change orientation in openimu.json.

closed 2022-11-11 by yiweisong